bewest/decoding-carelink #66: Battery change corrupts data log.

bustavo, 2015-03-22:
I think this is related with the Flipping page data corruption issue.

When the pump ( 754 ) goes out of battery and asks to change it, once you do it, and there is a "SensorSync", you get:

@@@ 19
{'raw': '0a00c3160f08'}
{'date': ''}
{'body': '0f08'}
[
  {
    "op": 10, 
    "name": "Could Not Decode", 
    "packet_size": 0
  }, 
  {
    "body": "", 
    "raw": "00-c4-16-0f", 
    "name": "SensorSync", 
    "date": "2015-03-22T00:04:00", 
    "date_type": "minSpecific", 
    "packet_size": 4, 
    "op": "0x0d"
  }
]

This is the list_cgm.py script on a ReadGlucoseHistory data file.

bewest, 2015-03-30:
Looks like `0x0a` should be some kind of `BatteryChange` record?

```
0a00c3160f08
0f08
```

What does carelink csv call these records?  Is it possible to get a binary page of data? (apologies if there is already one of these, hard to keep track of).

bustavo, 2015-03-30:
I am expecting to have a battery change between today & tomorrow. Will test and let you know!

bewest, 2015-03-30:
This is looking quite quite good so far, I was able to finally decode all my data.

Turns out my records basically ONLY include these records, and I just changed my battery.
(I have sensor enabled, but do not wear one.)

https://gist.github.com/bewest/c2a4c77b1f02164d9d2f

bewest, 2015-03-30:
This seems to fix: https://github.com/bewest/decoding-carelink/pull/72

bustavo, 2015-03-31:
@bewest It worked! :)

closed 2015-03-31 by bustavo