bewest/decoding-carelink #74: Setup for New User

atomspring, 2015-04-04:
Hello @bewest, I'm very proud of the work you've done thus far, thank you from a Type I owner of a Minimed 523!

I received a never-used carelink recently, and run Ubuntu 14.04 with 3.16 kernel, trying to connect my pump for the first time.

Can't seem to get it to go, I've installed everything but cannot get past "python decocare/stick.py /dev/ttyUSB0" as you can see in the captured log here: http://paste.ubuntu.com/10738995/

I'm noticing a lot of these in syslog: "kernel: [57091.695561] generic ttyUSB0: usb_serial_generic_write_bulk_callback - urb stopped: -32" perhaps these are related?

bewest, 2015-04-04:
On 3.16, try unplugging and replugging a few times.
On 3.16 you still need to run `./insert.sh`, and there might be some further hints in the `/var/log/syslog`.

bewest, 2015-04-05:
Pro-tip: `python -m decocare.scan` will auto-detect your carelink stick, and a usable path to the device file (or error if non).

`python -m decocare.stick` can be run from anywhere: `python -m decocare.stick $(python -m decocare.scan)` is a command I often use in your situation to figure out if the stick is "working."