bewest/decoding-carelink #76: Output bolus wizard settings to json

scottleibrand, 2015-04-05:
Could we add bolus wizard settings like sensitivity, bg_target_low, and bg_target_high to the --settings-out output?  Alternately, we could add another --bolus-settings-out type flag.  Seems like they really belong in --settings-out though.

bewest, 2015-04-05:
Also, the pump bolusing status.

scottleibrand, 2015-04-05:
Yeah, we probably need a new --status-out flag for bolusing/suspend status.  Might make sense to include IOB (unabsorbed_insulin_total) in with status as well, or do a third output flag for it.

bewest, 2015-05-04:
Is this still missing?  I want to get this implemented in @openaps quickly.

bewest, 2015-06-15:
I think next release goes long way toward exporting this to openaps, but would be good to review.

bewest, 2015-06-16:
This is now in `openaps 0.0.2`.

See https://github.com/bewest/openaps-example for examples:

* [bg-targets](https://github.com/bewest/openaps-example/blob/master/bg-targets.json)
* [carb ratios](https://github.com/bewest/openaps-example/blob/master/carb-ratios.json)
* [active basal profile](https://github.com/bewest/openaps-example/blob/master/current-basal-profile.json)
* [insulin sensitivities](https://github.com/bewest/openaps-example/blob/master/sensitivities.json)
* [misc settings](https://github.com/bewest/openaps-example/blob/master/settings.json)

I think that covers everything!  Closing for now, re-open or file new issue if/when needed. :smile_cat:

closed 2015-06-16 by bewest