bewest/decoding-carelink #80: Minimed Revel 723 compatibility

HellHarvest, 2015-04-16:
Has anyone tried this yet with this model (MMT-723NAS) of pump?  I'm trying what I can with mine, so far almost nothing works past insert.sh.  The serial number on my pump isn't just an integer.  It's PAR801601H, according to the sticker on the back of my pump.  Is there a simple command line tool that can use my carelink to get this kind of info?

loudnate, 2015-04-16:
Hi,

Your serial is just the numeric portion: 801601. 

Reading settings, history, and CGM values should work fine with your pump, among others. Setting boluses and basals, and key presses only work in "PC Connect" model if you have a U.S. model running firmware version 2.5 or greater. 

Please share your experiences communicating with the x23 models if you find anything to be different than above. 

Thanks!

HellHarvest, 2015-04-21:
I'm running firmware version 2.3A (according to the screen when I remove and reinsert the battery).  Anyone have any idea how to upgrade the firmware?

If not, then I'm in the market for a 722.

bewest, 2015-04-21:
Sorry, what errors are you getting, and what did you try?

You probably cannot send temp basal commands, but you should be able to read data from the pump just fine.

If you've got decocare installed, you should be able to run `python -m decocare.scan`, and `python -m decocare.stick $(python -m decocare.scan)` and get something interesting.  These commands talk to the stick, in order to diagnose if there's anything wrong with talking to the stick itself (this makes no attempt to communicate with the pump).

loudnate, 2015-04-22:
To clarify my earlier comment, version 2.3 should work fine. U.S. version
2.5 restricted changing settings to a special mode.

HellHarvest, 2015-04-22:
So a US-bought Minimed 723 with firmware 2.3 should work, as far as closing the loop and sending basal/bolus commands?

I want my pump to be able to receive basal/bolus commands from carelink.  If my pump can't do that, then I'm in the market for a 722.

Also, any 722 firmware I should watch out for while I'm in the market for one?

HellHarvest, 2015-04-22:
solderblob@ohm:~/decoding-carelink$ sudo ./remove.sh
[sudo] password for solderblob: 

solderblob@ohm:~/decoding-carelink$ sudo dmesg | grep ttyUSB
[  417.988573] usb 5-1: generic converter now attached to ttyUSB0
[593438.487347] generic ttyUSB0: generic converter now disconnected from ttyUSB0

solderblob@ohm:~/decoding-carelink$ sudo ./insert.sh

solderblob@ohm:~/decoding-carelink$ sudo dmesg | grep ttyUSB
[  417.988573] usb 5-1: generic converter now attached to ttyUSB0
[593438.487347] generic ttyUSB0: generic converter now disconnected from ttyUSB0
[593481.445017] usb 5-1: generic converter now attached to ttyUSB0

solderblob@ohm:~/decoding-carelink$ python decocare/stick.py /dev/ttyUSB0
INFO:__main__:howdy! I'm going to take a look at your carelink usb stick.
Traceback (most recent call last):
  File "decocare/stick.py", line 901, in <module>
    stick = Stick(link.Link(port))
  File "/home/solderblob/decoding-carelink/decocare/link.py", line 18, in __init__
    self.open( port )
  File "/home/solderblob/decoding-carelink/decocare/link.py", line 27, in open
    self.serial = serial.Serial( self.port, **kwds )
  File "/usr/lib/python2.7/dist-packages/serial/serialutil.py", line 261, in __init__
    self.open()
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 278, in open
    raise SerialException("could not open port %s: %s" % (self._port, msg))
serial.serialutil.SerialException: could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'

solderblob@ohm:~/decoding-carelink$ sudo python decocare/stick.py /dev/ttyUSB0
INFO:__main__:howdy! I'm going to take a look at your carelink usb stick.
INFO:link:Link opened serial port: Serial<id=0x7f7822db8910, open=True>(port='/dev/ttyUSB0', baudrate=9600, bytesize=8, parity='N', stopbits=1, timeout=0.5, xonxoff=False, rtscts=False, dsrdtr=False)
INFO:__main__:PROCESS:OPEN:0.017
INFO:__main__:PROCESS:START:0.094:ProductInfo:0x04
INFO:__main__:link Stick transmit[None] reader[None] download_i[False] status[None] poll_size[None] poll_i[None] command[<ProductInfo:0x04:size(64)>] processing ProductInfo:0x04)
INFO:__main__:send_force_read: attempt 0/2 send command, read until we get something within some timeout
INFO:__main__:link Stick transmit[None] reader[None] download_i[False] status[None] poll_size[None] poll_i[None] command[<ProductInfo:0x04:size(64)>] sending ProductInfo:0x04)
INFO:root:usb.write.len: 3
0000   0x04 0x00 0x00                             ...
DEBUG:__main__:sleeping 0.001
INFO:root:usb.read.len: 64
INFO:root:usb.read.raw:
0000   0x01 0x55 0x00 0x0a 0x14 0x00 0x00 0x00    .U......
0008   0x00 0x43 0x6f 0x6d 0x4c 0x69 0x6e 0x6b    .ComLink
0010   0x20 0x49 0x49 0x01 0x10 0x02 0x00 0x01     II.....
0018   0x01 0x03 0x00 0x00 0x00 0x00 0x00 0x00    ........
0020   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0028   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0030   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0038   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
INFO:__main__:quit send_force_read, found len: 64 expected 64 after 0 attempts
INFO:__main__:finished processing ProductInfo:0x04, {'description': 'ComLink II', 'software.version': '1.16', 'interfaces': [(0, 'Paradigm RF'), (1, 'USB')], 'product.version': '0.0', 'rf.freq': '916.5Mhz', 'serial': '0a1400'}
INFO:__main__:PROCESS:END:3.279:ProductInfo:0x04
INFO:__main__:{'description': 'ComLink II', 'software.version': '1.16', 'interfaces': [(0, 'Paradigm RF'), (1, 'USB')], 'product.version': '0.0', 'rf.freq': '916.5Mhz', 'serial': '0a1400'}
INFO:__main__:PROCESS:START:3.407:ProductInfo:0x04
INFO:__main__:link Stick transmit[None] reader[None] download_i[False] status[None] poll_size[None] poll_i[None] command[<ProductInfo:0x04:size(64)>] processing ProductInfo:0x04)
INFO:__main__:send_force_read: attempt 0/2 send command, read until we get something within some timeout
INFO:__main__:link Stick transmit[None] reader[None] download_i[False] status[None] poll_size[None] poll_i[None] command[<ProductInfo:0x04:size(64)>] sending ProductInfo:0x04)
INFO:root:usb.write.len: 3
0000   0x04 0x00 0x00                             ...
DEBUG:__main__:sleeping 0.001
INFO:root:usb.read.len: 64
INFO:root:usb.read.raw:
0000   0x01 0x55 0x00 0x0a 0x14 0x00 0x00 0x00    .U......
0008   0x00 0x43 0x6f 0x6d 0x4c 0x69 0x6e 0x6b    .ComLink
0010   0x20 0x49 0x49 0x01 0x10 0x02 0x00 0x01     II.....
0018   0x01 0x03 0x00 0x00 0x00 0x00 0x00 0x00    ........
0020   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0028   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0030   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
[... 514 lines cut ...]
0000   0x01 0x55 0x00 0x00 0x00 0x00 0x00 0x00    .U......
0008   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0010   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0018   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0020   0x00 0x01 0x00 0x4f 0x00 0x00 0x00 0x00    ...O....
0028   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0030   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0038   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
INFO:__main__:quit send_force_read, found len: 64 expected 64 after 0 attempts
INFO:__main__:finished processing RadioStats:0x05 0x00, {'errors.timeouts': 0, 'packets.transmit': 0L, 'errors.naks': 0, 'errors.sequence': 0, 'packets.received': 0L, 'errors.crc': 0}
INFO:__main__:PROCESS:END:1082.251:RadioStats:0x05 0x00
INFO:__main__:INTERFACE STATS:
{'radio': {'errors.crc': 0,
           'errors.naks': 0,
           'errors.sequence': 0,
           'errors.timeouts': 0,
           'packets.received': 0L,
           'packets.transmit': 0L},
 'usb': {'errors.crc': 3,
         'errors.naks': 5,
         'errors.sequence': 0,
         'errors.timeouts': 4,
         'packets.received': 377L,
         'packets.transmit': 385L}}
INFO:__main__:howdy! all done looking at the stick

HellHarvest, 2015-04-22:
Also, "export SERIAL=..." gives me no output.

solderblob@ohm:~/decoding-carelink$ export SERIAL=801601
solderblob@ohm:~/decoding-carelink$

bewest, 2015-04-22:
Looks like everything is working as expected, eg:

```
# this has no output, it's setting an environment variable
export SERIAL=801601
sudo mm-send-comm.py --init sleep 0 # should send power control on, and print model number
sudo mm-latest.py 30 # will get all status/activity for last 30 minutes
```

bewest, 2015-04-22:
I don't know that the version number of the firmware matters.
More specifically, european pumps and US pumps are different, US pumps have fewer features.  For sending bolus/basal commands, these were stripped out of US pumps while they were still being sold overseas, I have no idea how a version number maps to those changes.

bewest, 2015-04-22:
You might have noticed the `80-medtronic-carelink.rules` file.
This file allows the system to detect when the stick has been plugged in, and assigns it the "group" permission of `plugdev`.  If your user is also a member of group `plugdev`, you will be able to use the tools without needing `sudo` access, it will just work for your user.

Here's my file:
```bash
$ cat /etc/udev/rules.d/80-medtronic-carelink.rules 

# udev
# inserting  ftdio_sio module with our vendor's parameters creates a ttyUSB0
# udev can only match one entry at a time, but this triggers additional add entries.
# 0a21 8001
ATTRS{idVendor}=="0a21", ATTRS{idProduct}=="8001", \
  ENV{ID_MM_DEVICE_IGNORE}="1"

#ACTION=="add", SUBSYSTEM=="usb",                  \
#  ATTR{idVendor}=="0a21", ATTR{idProduct}=="8001", \
#  RUN+="/sbin/modprobe usbserial vendor=0x0a21 product=0x8001"

# create a symlink to the ttyUSB device created automatically above
ACTION=="add",                                       \
  SUBSYSTEM=="tty", \
  ATTRS{idVendor}=="0a21", ATTRS{idProduct}=="8001", \
  GROUP="plugdev" \
  SYMLINK+="ttyUSB.Carelink%n"

#ACTION=="remove",         \
#  SUBSYSTEM=="usb",        \
#  ENV{ID_MODEL_ID}=="0a21", \
#  ENV{ID_VENDOR_ID}=="8001", \
#  RUN+="/sbin/modprobe -r usbserial"

```

If you uncomment the modeprobe sections, it should also eliminate the need to run `./insert.sh` and `./remove.sh`... generally, FWIW, `./insert.sh` only needs to be run once, until the machine is rebooted, I believe.

FWIW, I strongly recommend against using these tools in sudo mode, I've seen this cause problems for people trying to update later on.

HellHarvest, 2015-04-23:
Thanks, I'll try to avoid sudo.  I'll try again later when I get home.

HellHarvest, 2015-05-06:
Toady I tried the following recommended actions, and got this output:

solderblob@anarchoPump ~/decoding-carelink $ python -m decocare.stick
usage:
/home/solderblob/decoding-carelink/decocare/stick.py /dev/ttyUSB0
solderblob@anarchoPump ~/decoding-carelink $ python -m decocare.scan
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/solderblob/decoding-carelink/decocare/scan.py", line 15, in <module>
    candidate = scan( )
  File "/home/solderblob/decoding-carelink/decocare/scan.py", line 12, in scan
    return results.pop( )
IndexError: pop from empty list

I could be wrong, but I feel like this kind of output is trying to tell me that something definitely went sideways.

bewest, 2015-05-06:
Setting your system up to talk to the usb stick is itself a challenge.

It's likely you don't have permission.  In order to get permission, try:

`sudo cp 80-medtronic-carelink.rules /etc/udev/rules.d/80-medtronic-carelink.rules`

Then `sudo udevadm control --reload`.
Now, when you [re]-insert the usb stick, you should be able to run the `./insert.sh` script (which also requires sudo), but hopefully the `python -m decocare.scan` will go well.

HellHarvest, 2015-05-06:
Should I call ./remove.sh before doing ./insert.sh?

solderblob@anarchoPump ~/decoding-carelink $ sudo cp 80-medtronic-carelink.rules /etc/udev/rules.d/80-medtronic-carelink.rules
solderblob@anarchoPump ~/decoding-carelink $ sudo udevadm control --reload

***I then physically removed carelink, then physically reinserted it **

solderblob@anarchoPump ~/decoding-carelink $ ./insert.sh
ERROR: could not insert 'usbserial': Module already in kernel
solderblob@anarchoPump ~/decoding-carelink $ sudo ./insert.sh
ERROR: could not insert 'usbserial': Module already in kernel
solderblob@anarchoPump ~/decoding-carelink $ python -m decocare.scan
/dev/serial/by-id/usb-0a21_8001-if00-port0

bewest, 2015-05-06:
Great, it's working.

You only need insert once per boot.

Try: `python -m decocare.stick $(python -m decocare.scan)` all in one line.

Or set your environment variable with something like `export SERIAL=112233` using your serial number, and try something like `mm-send-comm.py --init sleep 0`, you should see lots of stuff and your model number printed.  After that you can try `mm-latest.py 30` or similar.

bewest, 2015-05-06:
You can join https://gitter.im/nightscout/intend-to-bolus for chatting about using decocare and openaps.

FWIW, if you are mainly interested in getting data out, you might prefer using https://github.com/openaps/openaps as a "frontend" app for decocare.

HellHarvest, 2015-05-07:
I'm interested in completely closing the loop once I'm confident.

Thanks for the pointers.  I'm not new to linux, but I am new to github and python.  Thanks for your tenacity in keeping up with the forum posts on your repo.

bewest, 2015-05-07:
No problem, it's my pleasure.  I highly recommend using @openaps now, it's more of an SDK, not an all-in-one thing.  Much like decocare, it only does exactly what you tell it, but the output is nicer.  The command line tools are definitely better and easier to use.  @openaps does everything decocare does, but also has support for dexcom.  It'll be easier for me to track what people want to do with decocare and make sure there's a nice tool to do it there.  This repo is mainly for decoding the protocol itself.

bewest, 2015-06-15:
Closing, but please re-open with more questions/updates. :-)

closed 2015-06-15 by bewest