bewest/decoding-carelink #83: analyze data for correctness

bewest, 2015-05-04:
@loudnate left us some great data to verify data decoding.

This must be a new pump or something, it came in blank, with lots of NoDelivery errors, and then it looks like everything changes.  There are some novel records to be found here, and looks like we'll verify decoding for many.

bewest, 2015-05-04:
## [record 24](https://github.com/bewest/decoding-carelink/blob/bewest/dev/analysis/578398/ReadHistoryData-page-10.markdown#record-24-changeutility-2015-04-13t144447-head2-body0-op0x63)

## [ChangeAlarmNotifyMode](https://github.com/bewest/decoding-carelink/blob/bewest/dev/analysis/578398/changes.csv#L22)

`4150	4/13/15	14:44:47	4/13/15 14:44:47																														ChangeAlarmNotifyMode	MODE=vibration, ACTION_REQUESTOR=pump	14913477664	53848595	1349	Paradigm 522
`
??

loudnate, 2015-05-04:
I've got some analysis info and a PR decoding "Ian0b" as "AlarmSensor" coming your way tonight. Stay tuned.

loudnate, 2015-05-04:
The 722 I picked up on Craigslist seems to have a bad CMOS backup battery. After swapping the AAA, each time it has entered "Error A21" mode with the system clock reset and the sensor data gone. I expect history disappears too; I will verify that soon.

bewest, 2015-05-04:
Ah, try moving the clock back to some date before 2006?

loudnate, 2015-05-04:
Interesting - before or after changing the battery?

bewest, 2015-05-04:
@kenstack and others, it'd be cool to compare these two lists, methodically checking each one:

* https://github.com/bewest/decoding-carelink/blob/bewest/dev/analysis/578398/decoded_details.csv
* https://github.com/bewest/decoding-carelink/blob/bewest/dev/analysis/578398/details-no-sensor.csv

Here's a list of differences: https://github.com/bewest/decoding-carelink/blob/bewest/dev/analysis/578398/gap.diff

loudnate, 2015-05-05:
PR #85 is a list of manual comparisons for pump 736868, which I used when building PR #84.

A few other things that I noticed:
- `UnabsorbedInsulin` `age` value isn't handling 8-bit overflows (will need to find an additional byte somewhere)
- `UnabsorbedInsulin` `curve` is incorrect (not sure why yet)
- `BolusWizardBolusEstimate` `carb_ratio` and `bg_target_high` were incorrect (not sure why yet)
- `BasalProfileStart` includes the `profile_index` value in the second header byte. Not sure if that's important to anybody.

bewest, 2015-05-05:
I now do something like this:

`mm-pretty-csv <Carelink.csv>` to get cleaner carelink csv.

To decode glucose pages, I use:
` mm-decode-history-page.py --collate --data glucose --model 522 ReadGlucoseHistory-page-6.data | json -a date name op` or similar

For history:
`mm-decode-history-page.py --collate --model 522 ReadHistoryData-page-10.data | json -a timestsamp _type _head _date _body` or similar.

The first two columns should be identical-ish/comparable in all the above cases, including carelink csv.
This is only useful for debugging/hacking decocare.

For people who just want the json exports, please consider using @openaps, which provides a nice app to drive this and other devices.