bewest/decoding-carelink #98: ReadCarbRatios512 returning incorrect data for MM522

loudnate, 2015-05-30:
```python
{
    "i": 10,  # Actual ratio value
    "x": 0,  # 0-based index
    "r": 23,  # Actual offset value for the next record, in 30 minute increments
    "ratio": 23,  # Actual offset value for the next record, in 30 minute increments
    "offset": 300  # i * 30 (not useful)
},
```

bewest, 2015-06-15:
Ouch... is this still wrong, should probably be fixed for next release.

loudnate, 2015-06-15:
Yeah I filed this as a reminder, I'm happy to commit a fix. If you know any
individuals with earlier models than a 522 (which I have) that can help
test this, feel free to tag them here!

scottleibrand, 2015-06-15:
I have a 515.

-Scott

bewest, 2015-06-16:
This is for 512 only?

bewest, 2015-06-16:
I just tried a patch to get towards this in https://github.com/bewest/decoding-carelink/pull/103.  Basically had those two values backwards, it looks like?

bewest, 2015-06-16:
Oh, are you saying there's also off by one error, causing mismatch?  Is there any option to set grams vs exchanges?

commit d7285bcd (2015-06-16): fix #98

Should fix issue #98.
For 512, apparently I had the i and r values reversed.
closed 2015-06-16 by bewest