bewest/decoding-carelink #99: Random isoformat error decoding page

Fokko, 2015-05-31:
Hi guys,

I am tinkering with the software and everything seems to work. You guys did an amazing job on this project, I never knew that there was such a library. Currently I am writing my own algorithm which gives me feedback to improve my therapy.

Only now, I only got an error when I try to decode page 57 from my Medtronic Paradigm 754:
```
Traceback (most recent call last):
  File "./list_cgm.py", line 46, in <module>
    app.main( )
  File "./list_cgm.py", line 40, in main
    self.records.extend(page.decode( ))
  File "/home/fokko/Desktop/decoding-carelink-fokko/decocare/cgm/__init__.py", line 209, in decode
    record.update(date=date.isoformat())
AttributeError: 'NoneType' object has no attribute 'isoformat'
```

I also got the raw-data file in the repository:
https://github.com/Fokko/decoding-carelink/blob/master/data/ReadGlucoseHistory-page-57.data

## do stuff with an insulin pump over RF
using ` Namespace(autoinit=False, begin=None, bytesPerRecord=None, command='tweak', descr=None, dryrun=False, effectTime=None, end=None, init=False, maxRecords=None, name=None, no_postlude=False, no_prelude=False, no_rf_prelude=False, other='ReadGlucoseHistory', page=57, params=None, port='/dev/ttyUSB0', postfix=None, prefix=None, prefix_path='./data/', save=True, saveall=False, serial='417540', session_life=10, verbose=None) `
```

```javascript
{'radio': {'errors.crc': 0,
           'errors.naks': 0,
           'errors.sequence': 2,
           'errors.timeouts': 7,
           'packets.received': 2079L,
           'packets.transmit': 2177L},
 'usb': {'errors.crc': 0,
         'errors.naks': 5,
         'errors.sequence': 0,
         'errors.timeouts': 0,
         'packets.received': 4510L,
         'packets.transmit': 4515L}}
```

### PUMP MODEL: `ReadPumpModel:size[64]:data:'754'`
<class 'decocare.commands.ReadGlucoseHistory'> {'page': 57}
WARNING:decocare.stick:Stick transmit[TransmitPacket:ReadGlucoseHistory:size[1024]:[page][57]:data[0]:] reader[ReadRadio:size:142] download_i[5] status[<LinkStatus:0x03:status:size=??LinkStatus:error:True:reason:[]:size(0)>] poll_size[0] poll_i[False] command[<LinkStatus:0x03:status:size=??LinkStatus:error:True:reason:[]:size(0)>]:download(attempts[5],expect[0],results[384]:data[0]):BAD AILING
WARNING:decocare.stick:Stick transmit[TransmitPacket:ReadGlucoseHistory:size[1024]:[page][57]:data[0]:] reader[ReadRadio:size:142] download_i[6] status[<LinkStatus:0x03:status:size=??LinkStatus:error:True:reason:[]:size(0)>] poll_size[0] poll_i[False] command[<LinkStatus:0x03:status:size=??LinkStatus:error:True:reason:[]:size(0)>]:download(attempts[6],expect[0],results[384]:data[0]):BAD AILING
ERROR:decocare.stick:size (14) is less than 64 and not 15, which may cause an error.
WARNING:decocare.stick:Stick transmit[TransmitPacket:ReadGlucoseHistory:size[1024]:[page][57]:data[0]:] reader[ReadRadio:size:14] download_i[8] status[<LinkStatus:0x03:status:size=??LinkStatus:error:True:reason:[]:size(0)>] poll_size[0] poll_i[False] command[<LinkStatus:0x03:status:size=??LinkStatus:error:True:reason:[]:size(0)>]:download(attempts[8],expect[0],results[384]:data[0]):BAD AILING
WARNING:decocare.stick:Stick transmit[TransmitPacket:ReadGlucoseHistory:size[1024]:[page][57]:data[0]:] reader[ReadRadio:size:14] download_i[9] status[<LinkStatus:0x03:status:size=??LinkStatus:error:True:reason:[]:size(0)>] poll_size[0] poll_i[False] command[<LinkStatus:0x03:status:size=??LinkStatus:error:True:reason:[]:size(0)>]:download(attempts[9],expect[0],results[384]:data[0]):BAD AILING
response: ReadGlucoseHistory:size[1024]:[page][57]:data[1024]:
hexdump:
```python
0000   0x6d 0x00 0x00 0x01 0x0f 0x19 0x4a 0x55    m.....JU
0008   0x10 0x13 0x61 0x56 0x59 0x5b 0x5b 0x13    ..aVY[[.
0010   0x59 0x5c 0x5c 0x59 0x5d 0x13 0x61 0x13    Y\\Y].a.
0018   0x6a 0x7a 0x00 0x00 0x01 0x0f 0x19 0x51    jz.....Q
0020   0x56 0x10 0x7a 0x7c 0x84 0x8a 0x8d 0x13    V.z|....
0028   0x94 0x13 0x99 0xa8 0xbb 0xbd 0xbe 0xc0    ........
0030   0xc1 0x00 0x00 0x01 0x0f 0x19 0x55 0x57    ......UW
0038   0x10 0xc2 0xbd 0x06 0x07 0x29 0x07 0x38    .....).8
0040   0x07 0x41 0x07 0x36 0x07 0x23 0x07 0x19    .A.6.#..
0048   0x07 0x10 0x07 0x0b 0x07 0x07 0x07 0xc4    ........
0050   0xc2 0xbb 0xae 0xa6 0xa2 0x9e 0x99 0x94    ........
0058   0x9c 0xa1 0x9e 0x9a 0x13 0x9b 0x9c 0x13    ........
0060   0x9e 0x9f 0x9f 0xa1 0xa3 0x9a 0x96 0x97    ........
0068   0x9a 0x9d 0x9e 0x99 0x96 0x99 0x97 0x93    ........
0070   0x98 0x94 0x91 0x90 0x8e 0x91 0x94 0x98    ........
0078   0x00 0x03 0x00 0x03 0x00 0x03 0x00 0x03    ........
0080   0x00 0x03 0x00 0x03 0x00 0x03 0x00 0x03    ........
0088   0x00 0x03 0x00 0x03 0x00 0x03 0x00 0x03    ........
0090   0x00 0x03 0x00 0x03 0x00 0x03 0x00 0x03    ........
0098   0x00 0x03 0x00 0x03 0x00 0x03 0x00 0x03    ........
00A0   0x00 0x03 0x00 0x03 0x00 0x03 0x00 0x03    ........
00A8   0x00 0x03 0x00 0x03 0x00 0x03 0x00 0x03    ........
[... 102 lines cut ...]
03E0   0x1c 0x6d 0x4a 0x10 0x71 0x75 0x79 0x7e    .mJ.quy~
03E8   0x82 0x83 0x84 0x83 0x84 0x85 0x85 0x84    ........
03F0   0x85 0x84 0x84 0x83 0x83 0x82 0x82 0x81    ........
03F8   0x80 0x0f 0x3c 0x5b 0x4c 0x08 0xa2 0x9c    ..<[L...
```
#### decoded:
```python
'0000   0x6d 0x00 0x00 0x01 0x0f 0x19 0x4a 0x55    m.....JU\n0008   0x10 0x13 0x61 0x56 0x59 0x5b 0x5b 0x13    ..aVY[[.\n0010   0x59 0x5c 0x5c 0x59 0x5d 0x13 0x61 0x13    Y\\\\Y].a.\n0018   0x6a 0x7a 0x00 0x00 0x01 0x0f 0x19 0x51    jz.....Q\n0020   0x56 0x10 0x7a 0x7c 0x84 0x8a 0x8d 0x13    V.z|....\n0028   0x94 0x13 0x99 0xa8 0xbb 0xbd 0xbe 0xc0    ........\n0030   0xc1 0x00 0x00 0x01 0x0f 0x19 0x55 0x57    ......UW\n0038   0x10 0xc2 0xbd 0x06 0x07 0x29 0x07 0x38    .....).8\n0040   0x07 0x41 0x07 0x36 0x07 0x23 0x07 0x19    .A.6.#..\n0048   0x07 0x10 0x07 0x0b 0x07 0x07 0x07 0xc4    ........\n0050   0xc2 0xbb 0xae 0xa6 0xa2 0x9e 0x99 0x94    ........\n0058   0x9c 0xa1 0x9e 0x9a 0x13 0x9b 0x9c 0x13    ........\n0060   0x9e 0x9f 0x9f 0xa1 0xa3 0x9a 0x96 0x97    ........\n0068   0x9a 0x9d 0x9e 0x99 0x96 0x99 0x97 0x93    ........\n0070   0x98 0x94 0x91 0x90 0x8e 0x91 0x94 0x98    ........\n0078   0x00 0x03 0x00 0x03 0x00 0x03 0x00 0x03    ........\n0080   0x00 0x03 0x00 0x03 0x00 0x03 0x00 0x03    ........\n0088   0x00 0x03 0x00 0x03 0x00 0x03 0x00 0x03    ........\n0090   0x00 0x03 0x00 0x03 0x00 0x03 0x00 0x03    ........\n0098   0x00 0x03 0x00 0x03 0x00 0x03 0x00 0x03    ........\n00A0   0x00 0x03 0x00 0x03 0x00 0x03 0x00 0x03    ........\n00A8   0x00 0x03 0x00 0x03 0x00 0x03 0x00 0x03    ........\n00B0   0x00 0x03 0x00 0x03 0x00 0x03 0x00 0x03    ........\n00B8   0x00 0x03 0x00 0x03 0x00 0x03 0x00 0x03    ........\n00C0   0x00 0x03 0x00 0x03 0xa9 0x8f 0x9a 0x71    .......q\n00C8   0x46 0x0e 0x01 0x03 0x01 0x03 0x01 0x03    F.......\n00D0   0x02 0x03 0x00 0x03 0x00 0x03 0x00 0x03    ........\n00D8   0xa9 0x0f 0x1a 0x59 0x47 0x0e 0x01 0x03    ...YG...\n00E0   0x01 0x03 0x01 0x03 0x58 0x47 0x1a 0x0f    ....XG..\n00E8   0x1a 0x67 0x47 0x0f 0x54 0x13 0x54 0x13    .gG.T.T.\n00F0   0x5b 0x13 0x5f 0x5c 0x59 0x54 0x51 0x51    [._\\YTQQ\n00F8   0x54 0x56 0x5a 0x5b 0x5b 0x5a 0x58 0x58    TVZ[[ZXX\n0100   0x57 0x56 0x51 0x53 0x53 0x13 0x4a 0x44    WVQSS.JD\n0108   0x44 0x44 0x00 0x00 0x01 0x0f 0x1a 0x72    DD.....r\n0110   0x49 0x10 0x46 0x00 0x00 0x01 0x0f 0x1a    I.F.....\n0118   0x7a 0x49 0x10 0x43 0x13 0x3c 0x13 0x38    zI.C.<.8\n0120   0x13 0x36 0x34 0x13 0x35 0x33 0x13 0x2f    .64.53./\n0128   0x13 0x2c 0x13 0x28 0x13 0x27 0x00 0x00    .,.(.\'..\n0130   0x01 0x0f 0x1a 0x76 0x4a 0x10 0x13 0x27    ...vJ..\'\n0138   0x26 0x25 0x26 0x2b 0x2f 0x31 0x31 0x31    &%&+/111\n0140   0x2d 0x29 0x29 0x27 0x27 0x2c 0x2e 0x13    -))\'\',..\n0148   0x2c 0x28 0x25 0x27 0x28 0x27 0x24 0x22    ,(%\'(\'$"\n0150   0x1f 0x1d 0x13 0x02 0x13 0x02 0x13 0x02    ........\n0158   0x13 0x02 0x13 0x02 0x13 0x02 0x0f 0x5a    .......Z\n0160   0x5c 0x4d 0x08 0x13 0x02 0x13 0x02 0x0f    \\M......\n0168   0x5a 0x66 0x4d 0x08 0x0f 0x5a 0x6c 0x4d    ZfM..ZlM\n0170   0x0b 0x00 0x00 0x01 0x0f 0x1a 0x4a 0x4f    ......JO\n0178   0x10 0x00 0x00 0x01 0x0f 0x1a 0x54 0x52    ......TR\n0180   0x10 0x00 0x00 0x01 0x0f 0x1a 0x6b 0x52    ......kR\n0188   0x10 0x00 0x00 0x01 0x0f 0x1a 0x6d 0x55    ......mU\n0190   0x10 0x0f 0x3a 0x49 0x56 0x0d 0x0f 0x5a    ..:IV..Z\n0198   0x56 0x56 0x0b 0x00 0x00 0x01 0x0f 0x1a    VV......\n01A0   0x66 0x56 0x10 0x0f 0x3a 0x66 0x56 0x0d    fV..:fV.\n01A8   0x0f 0x5a 0x73 0x56 0x0b 0x00 0x00 0x01    .ZsV....\n01B0   0x0f 0x1a 0x5c 0x57 0x10 0x00 0x00  [... cut ...] 1    ........\n03F8   0x80 0x0f 0x3c 0x5b 0x4c 0x08 0xa2 0x9c    ..<[L...'
```
### end stats

```javascript
{'radio': {'errors.crc': 0,
           'errors.naks': 0,
           'errors.sequence': 2,
           'errors.timeouts': 10,
           'packets.received': 2104L,
           'packets.transmit': 2206L},
 'usb': {'errors.crc': 0,
         'errors.naks': 5,
         'errors.sequence': 0,
         'errors.timeouts': 0,
         'packets.received': 4609L,
         'packets.transmit': 4614L}}
```

loudnate, 2015-05-31:
Can you check against your carelink csv at the problematic record? It looks like the last know record was at time `2015-05-26T06:49:00`. What comes next? Anything unusual?
```(Pdb) record
{'raw': '07-10-07-19', 'date_type': 'minSpecific', 'op': '0x0b', 'name': 'SensorStatus', 'packet_size': 4}
(Pdb) len(records)
590
(Pdb) records[-1]
{'op': 7, 'name': 'Could Not Decode', 'packet_size': 0}
(Pdb) records[-2]
{'op': 7, 'name': 'Could Not Decode', 'packet_size': 0}
(Pdb) records[-3]
{'op': 7, 'name': 'Could Not Decode', 'packet_size': 0}
(Pdb) records[-4]
{'body': 'a9', 'raw': '46-71-9a-8f-a9', 'amount': 169, 'name': 'CalBGForGH', 'date': '2015-05-26T06:49:00', 'date_type': 'minSpecific', 'packet_size': 5, 'op': '0x0e'}
```
I'm not sure, but the `Could Not Decode` records, along with the `BAD AILING` entries in your ticket description, might suggest the data page is simply corrupt. You might want to try reading that page again.

bewest, 2015-05-31:
Curious how you got this page?  It looks so weird.
What did you get for `ReadCurGlucosePageNumber`?

Fokko, 2015-05-31:
Hi Guys,

Tomorrow I have access to a Windows computer, I will upload them then. The `ReadCurGlucosePageNumber` is 59 pages in total. What makes it strange is that pages 58-59 seem to  work fine. I'll post an update tomorrow.

bewest, 2015-05-31:
The glucose pages are not centered, and do not start from 0.

EG, mine looks like this:

```
{
  "isig": 32, 
  "page": 129, 
  "glucose": 32
}
```

For me, this means that the pages end at 129, and the glucose pages are the 32 pages before 129: pages `97-129`.  For me, asking for pages outside this range results in error.

bewest, 2015-06-02:
See https://github.com/bewest/decoding-carelink/pull/100, this allow much more parsing:  https://gist.github.com/bewest/6f35100a529085c32ac4

Fokko, 2015-06-02:
Thank you @bewest for the explanation for the pages, helps a lot.

I have extracted the data from Carelink, I posted it here:
http://www.sharecsv.com/s/3cc4c5c0cadad8ea131edabbce9da5f6/Carelink-reduced.csv

The problem might be the low battery, that was around the time that the battery of the sensor was getting pretty low. I recall charging it that day. Furthermore I am able now to download the CGM values using `ReadGlucoseHistory` and decode them using `list_cgm.py`.

Fokko, 2015-06-02:
Please let me know if I can supply additional information. Currently I am able to download the CGM values using `mm-send-comm.py` and `list_cgm.py`, but the `mm-latest.py` isn't working and gives `BAD AILING`.

Could it be that the 754 pump, which is not provided in the `models.py`, is not fully supported yet?

Fokko, 2015-06-02:
Seems to be working now:

## do stuff with an insulin pump over RF
using ` Namespace(autoinit=False, basal=True, basals=<open file '<stdout>', mode 'w' at 0x7f3b73c22150>, clock=True, dryrun=False, init=False, minutes=30, no_postlude=False, no_prelude=False, no_rf_prelude=False, parsed_data=<open file '<stdout>', mode 'w' at 0x7f3b73c22150>, port='', reservoir=True, reservoir_archive=<open file '<stdout>', mode 'w' at 0x7f3b73c22150>, rtc_archive=<open file '<stdout>', mode 'w' at 0x7f3b73c22150>, serial='417540', session_life=10, settings=<open file '<stdout>', mode 'w' at 0x7f3b73c22150>, status=True, temp=True, tempbasal=<open file '<stdout>', mode 'w' at 0x7f3b73c22150>, timezone=tzfile('/etc/localtime'), verbose=None) `
```
```javascript
{'radio': {'errors.crc': 0,
           'errors.naks': 0,
           'errors.sequence': 0,
           'errors.timeouts': 24,
           'packets.received': 0L,
           'packets.transmit': 24L},
 'usb': {'errors.crc': 1,
         'errors.naks': 0,
         'errors.sequence': 0,
         'errors.timeouts': 0,
         'packets.received': 847L,
         'packets.transmit': 848L}}
```
```
ERROR:decocare.stick:size (0) is less than 64 and not 15, which may cause an error.
WARNING:decocare.stick:Stick transmit[TransmitPacket:ReadPumpModel:data:unknown] reader[ReadRadio:size:0] download_i[2] status[<LinkStatus:0x03:status:size=??LinkStatus:error:True:reason:[]:size(0)>] poll_size[0] poll_i[False] command[<LinkStatus:0x03:status:size=??LinkStatus:error:True:reason:[]:size(0)>]:download(attempts[2],expect[0],results[0]:data[0]):BAD AILING
ERROR:decocare.stick:size (14) is less than 64 and not 15, which may cause an error.
WARNING:decocare.stick:Stick transmit[TransmitPacket:ReadPumpModel:data:unknown] reader[ReadRadio:size:14] download_i[4] status[<LinkStatus:0x03:status:size=??LinkStatus:error:True:reason:[]:size(0)>] poll_size[0] poll_i[False] command[<LinkStatus:0x03:status:size=??LinkStatus:error:True:reason:[]:size(0)>]:download(attempts[4],expect[0],results[0]:data[0]):BAD AILING
WARNING:decocare.stick:Stick transmit[TransmitPacket:ReadPumpModel:data:unknown] reader[ReadRadio:size:14] download_i[5] status[<LinkStatus:0x03:status:size=??LinkStatus:error:True:reason:[]:size(0)>] poll_size[0] poll_i[False] command[<LinkStatus:0x03:status:size=??LinkStatus:error:True:reason:[]:size(0)>]:download(attempts[5],expect[0],results[0]:data[0]):BAD AILING
WARNING:decocare.stick:Stick transmit[TransmitPacket:ReadPumpModel:data:unknown] reader[ReadRadio:size:14] download_i[6] status[<LinkStatus:0x03:status:size=??LinkStatus:error:True:reason:[]:size(0)>] poll_size[0] poll_i[False] command[<LinkStatus:0x03:status:size=??LinkStatus:error:True:reason:[]:size(0)>]:download(attempts[6],expect[0],results[0]:data[0]):BAD AILING
```
### PUMP MODEL: `ReadPumpModel:size[64]:data:'754'`

response: ReadSettings:size[64]:data:
```json
{'low_reservoir_warn_point': 20, 'keypad_lock_status': 0, 'maxBasal': 1600, 'temp_basal': {'percent': 100, 'type': 'Units/hour'}, 'low_reservoir_warn_type': 0, 'insulinConcentration': 50, 'audio_bolus_enable': False, 'variable_bolus_enable': False, 'alarm': {'volume': 2, 'mode': 2}, 'rf_enable': False, 'auto_off_duration_hrs': 0, 'block_enable': False, 'timeformat': 80, 'insulin_action_curve': 2, 'audio_bolus_size': 0, 'selected_pattern': 0, 'patterns_enabled': False, 'maxBolus': 0.0, 'paradigm_enabled': 1}
```
hexdump:
```python
0000   0x00 0x02 0x00 0x01 0x00 0x00 0xfa 0x00    ........
0008   0x50 0x01 0x00 0x00 0x00 0x00 0x00 0x64    P......d
0010   0x01 0x02 0x00 0x14 0x00 0x64 0x00 0x01    .....d..
0018   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0020   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0028   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0030   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0038   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
```

#### decoded:
```python
{
  "low_reservoir_warn_point": 20, 
  "keypad_lock_status": 0, 
  "maxBasal": 1600, 
  "temp_basal": {
    "percent": 100, 
    "type": "Units/hour"
  }, 
  "low_reservoir_warn_type": 0, 
  "insulinConcentration": 50, 
  "audio_bolus_enable": false, 
  "variable_bolus_enable": false, 
  "alarm": {
    "volume": 2, 
    "mode": 2
  }, 
  "rf_enable": false, 
  "auto_off_duration_hrs": 0, 
  "block_enable": false, 
  "timeformat": 80, 
  "insulin_action_curve": 2, 
  "audio_bolus_size": 0, 
  "selected_pattern": 0, 
[... 282 lines cut ...]
  * should quit True True
  * found record SensorAlert 2015-06-01T22:49:11
  * should quit True True
  * found record BGReceived 2015-06-01T22:44:24
  * should quit True True
  * found record CalBGForPH 2015-06-01T22:44:24
  * should quit True True
```json
[]
```
### end stats
```javascript
{'radio': {'errors.crc': 0,
           'errors.naks': 0,
           'errors.sequence': 0,
           'errors.timeouts': 28,
           'packets.received': 24L,
           'packets.transmit': 53L},
 'usb': {'errors.crc': 1,
         'errors.naks': 0,
         'errors.sequence': 0,
         'errors.timeouts': 0,
         'packets.received': 984L,
         'packets.transmit': 985L}}
```

bewest, 2015-06-15:
I think we solved this, not sure.

bewest, 2015-06-16:
Closing, this shipped in master and pypi. :-)

closed 2015-06-16 by bewest