bewest/decoding-carelink #116: Not able to set temp basal >=6.4U/hr in decocare

TC2013, 2015-07-06:
As @loudnate explained, "...due to how basal rates are encoded (# of 1/40 U doses) and that 6.4 * 40 = 256, which is the maximum integer that could be communicated with one byte."

TC2013, 2016-09-07:
Loop/Loopkit users have reported being able to set temp basal rates up to 35.0u/hr.

TC2013, 2016-09-13:
I tested this with loop 0.12.1beta running on an iPhone 6 and a 722 pump and it was not able to set temp basals > 6.35u/hr.

ps2, 2016-09-25:
Here is how it's done in MinimedKit: https://github.com/ps2/rileylink_ios/blob/dev/MinimedKit/Messages/ChangeTempBasalCarelinkMessageBody.swift#L21

ps2, 2016-09-25:
Also, reading current temp basal rate supports up to 34U/hr as well:

https://github.com/ps2/rileylink_ios/blob/dev/MinimedKit/Messages/ReadTempBasalCarelinkMessageBody.swift#L35

Bender1061, 2016-09-30:
So this is why. I ran into this when I tried to change my max rate on my pump from 6 to 7U/h. Which would be most useful since I'm extremely insulin resistant due to the steroids I'm on.

scottleibrand, 2016-10-18:
@Bender1061 or @TC2013 can you test this fix?  Let us know if you need help installing @ecc1's branch.

Bender1061, 2016-10-18:
@scottleibrand  yes I could use some help on that. And I'm currently not using Decocare at all. So how would this integrate with my loop?

SandraK82, 2016-10-20:
@ecc1 @scottleibrand there is still at least one error remaing after >=6.4U fix from @ecc1:
reporting enact/suggested.json
{"temp":"absolute","bg":145,"tick":"+5","eventualBG":141,"snoozeBG":141,"predBGs":{"IOB":[..shortend..],"aCOB":[..shortend..],"COB":..shortend..]},"COB":5,"IOB":3.42,"reason":"COB: 5, Dev: 69, BGI: -6.51, ISF: 21, Target: 80; Eventual BG 141 >= 85, temp 6.325<7.5U/hr","duration":30,"rate":7.5}
rm: cannot remove ‘enact/enacted.json’: No such file or directory
enact/enacted.json  raised  byte must be in range(0, 256)
pump://JSON/set_temp_basal/enact/enacted.json

scottleibrand, 2016-10-20:
@Bender1061 you can clone the branch and then install it with setup.py develop, similar to how it's done here: https://github.com/openaps/docs/blob/master/scripts/quick-src.sh#L11-L15

closed 2016-10-26 by bewest