bewest/decoding-carelink #124: CalBGForGH needs another byte/bit

loudnate, 2015-08-11:
```
{
    "body": "2a", 
    "packet_size": 5, 
    "amount": 42, 
    "name": "CalBGForGH", 
    "raw": "86-20-2b-8f-2a", 
    "date": "2015-08-11T06:32:00", 
    "date_type": "minSpecific", 
    "_tell": 81, 
    "op": "0x0e"
  }
```

Was actually `298`. (Nevermind that the 1st-gen Medtronic sensor was reporting 180...)

oskarpearson, 2015-09-09:
I think I *may* have encountered this issue too, but I'm not 100% sure given the description :) 

The pump was disconnected,  and when reconnected BG was high. The sensor was then re-linked, and calibrated to 16.7 (around 300 - I only have UK units).  4.5 units insulin was given.

The screen of the pump showed BG of 16.7 a couple minutes later.

```
  {
    "body": "2d",
    "packet_size": 5,
    "amount": 45,
    "name": "CalBGForGH",
    "raw": "93-53-29-0f-2d",
    "date": "2015-09-09T19:19:00",
    "date_type": "minSpecific",
    "_tell": 45,
    "op": "0x0e"
  },
```

loudnate, 2015-09-09:
45 is 2D in hex. 12D in hex is 301. We need to find the bit that is flipped
in this entry but not flipped in entries less than 256 mg/dL. Can you find
another data point?

loudnate, 2015-09-23:
The third left-most bit in the 3rd byte (day) is the magic one.

PR #132 (2015-09-23): Adding a high bit for calibration glucose
Fixes #124
commit e7291f3b (2015-09-23): Adding a high bit for calibration glucose

Fixes #124
closed 2015-10-20 by loudnate