bewest/decoding-carelink #129: Opcode 50

loudnate, 2015-09-08:
Creates a hiccup in the history page. Carelink CSV to follow.

```
{
    "_type": "Ian50", 
    "_description": "Ian50 2015-09-07T08:57:33 head[2], body[34] op[0x50]", 
    "timestamp": "2015-09-07T08:57:33", 
    "_body": "200128005a1400283c212562b44600210128005a1400283c212562b446000b6b0080", 
    "appended": [
      {
        "_body": "", 
        "_head": "7a28", 
        "_type": "Base", 
        "_date": "a70f5b0094", 
        "_description": "Base (2004, 8, 0, 27, 15, 39) head[2], body[0] op[0x7a]"
      }, 
      {
        "_body": "", 
        "_head": "4b09", 
        "_type": "Base", 
        "_date": "670f105000", 
        "_description": "Base (2000, 4, 16, 16, 15, 39) head[2], body[0] op[0x4b]"
      }, 
      {
        "_body": "", 
        "_head": "6428", 
        "_type": "ChangeTimeDisplay", 
        "_date": "5a00004000", 
        "_description": "ChangeTimeDisplay (2000, 4, 0, 0, 0, 26) head[2], body[0] op[0x64]"
      }
    ], 
    "_head": "5000", 
    "_date": "a17908070f"
  },
```

loudnate, 2015-09-08:
This was turning the sensor back on:
```
19,9/7/15,08:57:16,9/7/15 08:57:16,,,,,,,,,,,,,Resume,,,,,,,,,,,,,,,,,ChangeSuspendState,"STATE=normal_pumping, ACTION_REQUESTOR=pump",15506617846,54028343,165,Paradigm Revel - 723
20,9/7/15,08:57:33,9/7/15 08:57:33,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,ChangeSensorSetup2,"NEW_CONFIG_DATUM_ID=15506617844, OLD_CONFIG_DATUM_ID=15506617843, ACTION_REQUESTOR=pump",15506617845,54028343,164,Paradigm Revel - 723
21,9/7/15,08:57:33,9/7/15 08:57:33,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,ChangeSensorSetupConfig2,"IS_SENSOR_ENABLED=true, BG_UNITS=mg dl, HIGH_AREA_UNDER_CURVE=180, HIGH_GLUCOSE_SNOOZE_TIME=5400000, LOW_AREA_UNDER_CURVE=70, LOW_GLUCOSE_SNOOZE_TIME=1200000, CAL_REMINDER_ENABLE=true, CAL_REMINDER_TIME=3600000, ALARM_SNOOZE_TIME=2400000, MISSED_DATA_TIME=2400000, IS_AUTO_CAL_ENABLE=false, TRANSMITTER_ID=2172258",15506617844,54028343,163,Paradigm Revel - 723
22,9/7/15,08:57:33,9/7/15 08:57:33,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,ChangeSensorSetupConfig2,"IS_SENSOR_ENABLED=false, BG_UNITS=mg dl, HIGH_AREA_UNDER_CURVE=180, HIGH_GLUCOSE_SNOOZE_TIME=5400000, LOW_AREA_UNDER_CURVE=70, LOW_GLUCOSE_SNOOZE_TIME=1200000, CAL_REMINDER_ENABLE=true, CAL_REMINDER_TIME=3600000, ALARM_SNOOZE_TIME=2400000, MISSED_DATA_TIME=2400000, IS_AUTO_CAL_ENABLE=false, TRANSMITTER_ID=2172258",15506617843,54028343,162,Paradigm Revel - 723
23,9/7/15,08:58:00,9/7/15 08:58:00,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,SensorStatus,STATUS_TYPE=lost,15506621074,54028343,393,Paradigm Revel - 723
24,9/7/15,08:58:00,9/7/15 08:58:00,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,SensorTimestamp,TIMESTAMP_TYPE=gap,15506621075,54028343,394,Paradigm Revel - 723
25,9/7/15,08:58:00,9/7/15 08:58:00,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,SensorError,"ERROR_TYPE=end, VCNTR=-1.73, BACKFILL_INDICATOR=false",15506621076,54028343,395,Paradigm Revel - 723
26,9/7/15,08:58:00,9/7/15 08:58:00,,,,,,,,,,,,,,,,,,,,,,,,,Sensor Alert: Sensor End (107),,,,,AlarmSensor,"ALARM_TYPE=107, AMOUNT=0, ACTION_REQUESTOR=sensor",15506617842,54028343,161,Paradigm Revel - 723
27,9/7/15,09:09:00,9/7/15 09:09:00,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,SensorSync,SYNC_TYPE=new,15506621073,54028343,392,Paradigm Revel - 723
28,9/7/15,09:11:20,9/7/15 09:11:20,,,,,,,Normal,1.6,1.6,,,,,,,,,,,,,,,,,,,,,BolusNormal,"AMOUNT=1.6, PROGRAMMED_AMOUNT=1.6, ACTION_REQUESTOR=paradigm link or b key, ENABLE=true, IS_DUAL_COMPONENT=false, UNABSORBED_INSULIN_TOTAL=0",15506617840,54028343,159,Paradigm Revel - 723
```

bewest, 2016-05-29:
This seems important.

bewest, 2016-05-29:
Any overlap with this one? https://github.com/bewest/decoding-carelink/issues/131