bewest/decoding-carelink #133: Fuser in dev branch

channemann, 2015-09-29:
New dev branch causing issues with stick being in use (same sequence of report invoking worked fine before merge):

```
pump://JSON/read_status/read_status.json
reporting read_status.json
pump://JSON/read_battery_status/read_battery_status.json
Traceback (most recent call last):
  File "/usr/local/bin/openaps-report", line 6, in <module>
    exec(compile(open(__file__).read(), __file__, 'exec'))
  File "/home/pi/openaps/bin/openaps-report", line 82, in <module>
    app( )
  File "/home/pi/openaps/openaps/cli/__init__.py", line 44, in __call__
    self.run(self.args)
  File "/home/pi/openaps/bin/openaps-report", line 75, in run
    output = app(args, self)
  File "/home/pi/openaps/openaps/cli/subcommand.py", line 50, in __call__
    return self.method.main(args, app)
  File "/home/pi/openaps/openaps/reports/invoke.py", line 40, in main
    output = task.method(args, app)
  File "/home/pi/openaps/openaps/uses/use.py", line 25, in __call__
    self.before_main(args, app)
  File "/home/pi/openaps/openaps/vendors/medtronic.py", line 59, in before_main
    self.setup_medtronic( )
  File "/home/pi/openaps/openaps/vendors/medtronic.py", line 117, in setup_medtronic
    self.uart = stick.Stick(link.Link(self.scanner( )))
  File "/home/pi/decoding-carelink/decocare/link.py", line 23, in __init__
    raise AlreadyInUseException("{port} already in use".format(port=port))
decocare.link.AlreadyInUseException: /dev/serial/by-id/usb-0a21_8001-if00-port0 already in use
```

channemann, 2015-10-06:
Solved with https://github.com/openaps/openaps/commit/2939939aa23d286122ef9b519477698339624133

bewest, 2015-10-07:
openaps has a bug, which it attempts to open the device multiple times.  Fixed in dev branch of openaps.

closed 2015-10-07 by bewest