bewest/decoding-carelink #142: Units issue (mmol/L vs mg/dL) in ReadInsulinSensitivities

oskarpearson, 2015-10-25:
Hi

We're based in the UK, so our pump (722) is configured with mmol/L instead of mg/dL.

On the pump's screen, the sensitivity value is currently "2.2" mmol/L per Unit. Calling ReadInsulinSensitivities returns a value of '22'. Something appears to be dropping the '.' and not keeping track of the units.

```
mm-send-comm.py  --serial REMOVED  tweak ReadInsulinSensitivities
## do stuff with an insulin pump over RF
using ` Namespace(autoinit=False, begin=None, bytesPerRecord=None, command='tweak', descr=None, dryrun=False, effectTime=None, end=None, init=False, maxRecords=None, name=None, no_postlude=False, no_prelude=False, no_rf_prelude=False, other='ReadInsulinSensitivities', page=None, params=None, port='', postfix=None, prefix=None, prefix_path='', save=False, saveall=False, serial='REMOVED', session_life=10, verbose=None) `
```
```
```
```
```
```
```
```
```javascript
{'radio': {'errors.crc': 0,
           'errors.naks': 0,
           'errors.sequence': 0,
           'errors.timeouts': 14,
           'packets.received': 162L,
           'packets.transmit': 182L},
 'usb': {'errors.crc': 0,
         'errors.naks': 0,
         'errors.sequence': 0,
         'errors.timeouts': 0,
         'packets.received': 771L,
         'packets.transmit': 771L}}
```
```
```
### PUMP MODEL: `ReadPumpModel:size[64]:data:'722'`
<class 'decocare.commands.ReadInsulinSensitivities'> {}
ERROR:decocare.stick:size (0) is less than 64 and not 15, which may cause an error.
response: ReadInsulinSensitivities:size[64]:data:{'sensitivities': [{'i': 0, 'start': datetime.time(0, 0), 'sensitivity': 22, 'offset': 0, 'x': 0}], 'action_type': 'Regular', 'first': 2}
hexdump:
```python
0000   0x02 0x00 0x16 0x00 0x00 0x00 0x00 0x00    ........
0008   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0010   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0018   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0020   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0028   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0030   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0038   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
```
#### decoded:
```python
{'sensitivities': [{'i': 0, 'start': datetime.time(0, 0), 'sensitivity': 22, 'offset': 0, 'x': 0}], 'action_type': 'Regular', 'first': 2}
```
### end stats
```
```
```javascript
{'radio': {'errors.crc': 0,
           'errors.naks': 0,
           'errors.sequence': 0,
           'errors.timeouts': 16,
           'packets.received': 164L,
           'packets.transmit': 186L},
 'usb': {'errors.crc': 0,
         'errors.naks': 0,
         'errors.sequence': 0,
         'errors.timeouts': 0,
         'packets.received': 809L,
         'packets.transmit': 809L}}
```

Oskar

loudnate, 2015-10-25:
I think the "first" param is indicative of the units. I will check my own data later.

bewest, 2015-10-25:
I only see first set to `1`, looks like `2` means `mmol/L` per unit.

I'd love to do this checking rigorously against the other types of data.  Would it make sense to create a checklist as a community?

bewest, 2015-10-25:
```
$ (cat <<EOF
0000   0x02 0x00 0x16 0x00 0x00 0x00 0x00 0x00    ........
0008   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0010   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0018   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0020   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0028   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0030   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0038   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
EOF
         ) | cut -d ' ' -f 3- | cut -d ' ' -f 1-8 | sed -e "s/ 0x//g" | while read line ; do echo ', "'$line'"'; done | tee t
, "02001600000000"
, "00000000000000"
, "00000000000000"
, "00000000000000"
, "00000000000000"
, "00000000000000"
, "00000000000000"
, "00000000000000"
bewest@hither:~/src/nightscout/cgm-remote-monitor$
```

bewest, 2015-10-25:
Indirectly related, but also affected: https://github.com/openaps/openaps/issues/65

oskarpearson, 2015-10-27:
@bewest I think we can close this given https://github.com/bewest/decoding-carelink/commit/c548d10b571f16517f7910a0cc0105ed7ef9218c#diff-7eb4cea574485b05ea765bdf71ccb4ab - right? Thanks for jumping in and working on it so fast! :+1:

closed 2021-03-22 by oskarpearson