bewest/decoding-carelink #152: Trouble getting /dev/ttyUSB0 to show up

jotomo, 2015-11-28:
Apologies if this isn't the right place for such a request, I couldn't find a better one so far.

I'm running Arch Linux with a 4.2.5 kernel. I see the attached _Contour Next Link_ USB device:
```
$ lsub
Bus 003 Device 011: ID 1a79:6210 Bayer Health Care LLC 
```
And I've adjusted the _insert.sh_ script to use the right ids:
```
sudo modprobe -first-time usbserial vendor=0x1a79 product=0x6210
```
When running the script, the log reads:
```
Nov 28 23:17:57 mal kernel: usbcore: registered new interface driver usbserial
Nov 28 23:17:57 mal kernel: usbcore: registered new interface driver usbserial_generic
Nov 28 23:17:57 mal kernel: usbserial: USB Serial support registered for generic
```
But no mention of anything related to _/dev/ttyUSB0_, which refuses to show up.

I'd be grateful for any pointers, as I'm out of ideas right now.

bewest, 2015-11-28:
The contour nextlink is a different device, see these for more info:

* https://github.com/bewest/decoding-bayer
* https://github.com/bewest/bayer-medtronic-interface

jotomo, 2015-11-28:
Thanks, that explains that :-) Now I can see a _/dev/hidrawX_ device popping up.
Reading the links you provided, I'm unsure whether it's currently possible to achieve what I have in mind: use the Contour NextLink to read the pumps (Medtronic 640g) history. The NextLink is bundled and acts as a hub for the pump.

bewest, 2015-11-28:
It should be possible to get 640g history.  You can use USBlyzer on Windows to create a sniff/capture the communication from Carelink webapp to the Nextlink device to see how they do it.

jotomo, 2015-11-28:
Cool. Thanks for the speedy help!

closed 2015-11-28 by jotomo