bewest/decoding-carelink #156: iter_pump issues with 0.0.19

tghoward, 2015-12-24:
upgraded yesterday to 0.0.19 from 0.0.18 and all was fine immediately, but now it is complaining. So I'm assuming that means there's a record that isn't getting parsed correctly or something.  iter_pump_hours gives the same error. read_history does give data.   Pump is a MM 723 with integrated cgm. 

`openaps use pump723 iter_pump 2
ERROR month must be in 1..12 0000   0x0c 0x00                                  ..
Traceback (most recent call last):
  File "/usr/local/bin/openaps-use", line 4, in <module>
    __import__('pkg_resources').run_script('openaps==0.0.8', 'openaps-use')
  File "/usr/local/lib/python2.7/dist-packages/setuptools-18.4-py2.7.egg/pkg_resources/__init__.py", line 735, in run_script
  File "/usr/local/lib/python2.7/dist-packages/setuptools-18.4-py2.7.egg/pkg_resources/__init__.py", line 1652, in run_script
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.8-py2.7.egg/EGG-INFO/scripts/openaps-use", line 63, in <module>
    app( )
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.8-py2.7.egg/openaps/cli/__init__.py", line 44, in __call__
    self.run(self.args)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.8-py2.7.egg/EGG-INFO/scripts/openaps-use", line 57, in run
    output = app(args, self)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.8-py2.7.egg/openaps/uses/__init__.py", line 92, in __call__
    return self.method.selected(args)(args, app)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.8-py2.7.egg/openaps/uses/__init__.py", line 31, in __call__
    return self.method(args, app)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.8-py2.7.egg/openaps/uses/use.py", line 45, in __call__
    output = self.main(args, app)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.8-py2.7.egg/openaps/vendors/medtronic.py", line 373, in main
    for rec in self.range( ):
  File "/usr/local/lib/python2.7/dist-packages/decocare-0.0.19-py2.7.egg/decocare/models/__init__.py", line 67, in __call__
    for record in page:
  File "/usr/local/lib/python2.7/dist-packages/decocare-0.0.19-py2.7.egg/decocare/models/__init__.py", line 46, in download_page
    for record in self.find_records(page):
  File "/usr/local/lib/python2.7/dist-packages/decocare-0.0.19-py2.7.egg/decocare/models/__init__.py", line 139, in find_records
    records = decoder.decode( )
  File "/usr/local/lib/python2.7/dist-packages/decocare-0.0.19-py2.7.egg/decocare/history.py", line 805, in decode
    record = parse_record(self.stream, B, larger=larger, model=self.model)
  File "/usr/local/lib/python2.7/dist-packages/decocare-0.0.19-py2.7.egg/decocare/history.py", line 737, in parse_record
    record.parse( head + date + body )
  File "/usr/local/lib/python2.7/dist-packages/decocare-0.0.19-py2.7.egg/decocare/records/base.py", line 66, in parse
    return self.decode( )
  File "/usr/local/lib/python2.7/dist-packages/decocare-0.0.19-py2.7.egg/decocare/history.py", line 697, in decode
    return (dict(valid_date=date(*mid).isoformat()))
ValueError: month must be in 1..12
`

tghoward, 2015-12-24:
In an attempt to make it more readable, here's a screenshot of the same error messages. 

[image]

Unfortunately, I'm about to leave on vacation for 1.5 weeks so I might be in radio silence for a while. Just wanted to get this report in.

tghoward, 2015-12-24:
Reverting back to 0.0.18 with 
`sudo easy_install decocare=0.0.18`
fixes the problem (e.g. iter_pump works again)

tghoward, 2015-12-24:
Whoops, looks like I already reported this, here:

https://github.com/openaps/openaps/issues/70

But now it was completely breaking the use or report commands.  Sorry for the redundancy.

tghoward, 2016-01-10:
I just placed more page dumps here:

https://github.com/tghoward/decoding-carelink/tree/thowdat/logs/2016-01-09

bewest, 2016-01-10:
## START /home/bewest/src/decocare/logs/2016-01-09/ReadHistoryData-page-31.data
#### STOPPING DOUBLE NULLS @ 1020, found 2 nulls
reading more to debug 0x00
    0000   0x00 0x00                                  ..
              0    0
##### DEBUG HEX
    0000   0xfe 0xe3                                  ..
##### DEBUG DECIMAL
            254  227
#### RECORD 0 TempBasal 2015-12-23T08:50:22 head[2], body[1] op[0x33]
###### DECODED
```python
{'rate': 0.0, 'temp': 'absolute'}
```
    op hex (2)
    0000   0x33 0x00                                  3.
    decimal
             51    0
    datetime (2015-12-23T08:50:22)
    0000   0xd6 0x32 0x08 0x57 0x0f                   .2.W.
    body (1)
    hex
    0000   0x00                                       .
    decimal
              0

#### RECORD 1 TempBasalDuration 2015-12-23T08:50:22 head[2], body[0] op[0x16]
###### DECODED
```python
{'duration (min)': 0}
```
    op hex (2)
    0000   0x16 0x00                                  ..
    decimal
             22    0
    datetime (2015-12-23T08:50:22)
    0000   0xd6 0x32 0x08 0x57 0x0f                   .2.W.
    body (0)

#### RECORD 2 BasalProfileStart 2015-12-23T08:50:23 head[2], body[3] op[0x7b]
###### DECODED
```python
{'offset': 27000000, 'profile_index': 2, 'rate': 1.25}
```
    op hex (2)
    0000   0x7b 0x02                                  {.
    decimal
            123    2
    datetime (2015-12-23T08:50:23)
    0000   0xd7 0x32 0x08 0x17 0x0f                   .2...
    body (3)
    hex
    0000   0x0f 0x32 0x00                             .2.
    decimal
             15   50    0

#### RECORD 3 TempBasal 2015-12-23T09:00:23 head[2], body[1] op[0x33]
###### DECODED
```python
{'rate': 0.675, 'temp': 'absolute'}
```
    op hex (2)
    0000   0x33 0x1b                                  3.
    decimal
             51   27
    datetime (2015-12-23T09:00:23)
    0000   0xd7 0x00 0x09 0x57 0x0f                   ...W.
    body (1)
    hex
    0000   0x00                                       .
[... 1302 lines cut ...]
    datetime ((2085, 0, 0, 19, 15, 24))
    0000   0x18 0x0f 0x33 0x00 0xd5                   ..3..
    body (0)

#### RECORD 101 Base (2000, 4, 22, 0, 15, 24) head[2], body[0] op[0x05]

    op hex (2)
    0000   0x05 0x08                                  ..
    decimal
              5    8
    datetime ((2000, 4, 22, 0, 15, 24))
    0000   0x58 0x0f 0x00 0x16 0x00                   X....
    body (0)

#### RECORD 102 Base (2000, 1, 0, 15, 24, 8) head[2], body[0] op[0xd5]

    op hex (2)
    0000   0xd5 0x05                                  ..
    decimal
            213    5
    datetime ((2000, 1, 0, 15, 24, 8))
    0000   0x08 0x58 0x0f 0x00 0x00                   .X...
    body (0)

`end /home/bewest/src/decocare/logs/2016-01-09/ReadHistoryData-page-31.data: 103 records`