bewest/decoding-carelink #165: Git clone 7/1/15 not working

gazhay, 2016-01-07:
Attempting to run stick.py /dev/ttyUSB0 having found that carelink is on /dev/ttyUSB0 from dmesg

```
INFO:__main__:howdy! I'm going to take a look at your carelink usb stick.
Traceback (most recent call last):
  File "stick.py", line 906, in <module>
    stick = Stick(link.Link(port))
  File "/home/gaz/.local/lib/python2.7/site-packages/decocare/link.py", line 24, in __init__
    self.open( port )
  File "/home/gaz/.local/lib/python2.7/site-packages/decocare/link.py", line 33, in open
    self.serial = serial.Serial( self.port, **kwds )
  File "/home/gaz/.local/lib/python2.7/site-packages/serial/serialutil.py", line 171, in __init__
    self.open()
  File "/home/gaz/.local/lib/python2.7/site-packages/serial/serialposix.py", line 311, in open
    self._update_dtr_state()
  File "/home/gaz/.local/lib/python2.7/site-packages/serial/serialposix.py", line 605, in _update_dtr_state
    fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_DTR_str)
IOError: [Errno 22] Invalid argument
```

gazhay, 2016-01-12:
Seems pyserial has changed, adding hardware flow.control bools to link.py seems to fix the issue.

codebudo, 2016-03-06:
I had this same issue and switching from the 'master' to 'dev' branches fixed the problem. I suggest the 'dev' branch should be merged to master ASAP to save others from falling in this trap (it cost me an hour).

Also, since this bug is fixed this issue should be closed.

closed 2016-03-07 by bewest