bewest/decoding-carelink #174: what is command for entering journal events?

bewest, 2016-03-06:
With @loudnate.

loudnate, 2016-03-06:
`233` triggers `ChangeWatchdogMarriageProfile` and `ChangeOtherDeviceID`
`237` triggers `questionable7c`

loudnate, 2016-03-06:
`242` triggers `ChangeCaptureEventEnable`

loudnate, 2016-03-06:
`mm-send-comm.py  ManualCommand 242 --params 1` --> CaptureEvent menu enabled
`mm-send-comm.py  ManualCommand 242 --params 0` --> CaptureEvent menu disabled

loudnate, 2016-03-06:
`181` records `questionable66`

loudnate, 2016-03-06:
Stay away from `179`. 
`179` `0 1` sets all the pixels black. Menus appear blank.
`179` `1 0` displays the rest screen again, but menus appear blank.

I haven't found a way out of this yet.

loudnate, 2016-03-06:
`179` sets the contrast. `195` reads it. I was able to read a correct contrast value (`2, 7`) and re-set the pump to make it visible again.

bewest, 2016-03-06:
Set contrast

loudnate, 2016-03-07:
`241` reads the status of `ChangeCaptureEventEnable`. `242` sets it.