bewest/decoding-carelink #181: mm-send-comm.py readhistorydata returns same bolus twice

kenstack, 2016-04-11:
strange - I dont see it often but this happens sometimes - I think I saw in @loudnate tools that he checks for repeats so perhaps this is a known issue?  If so feel free to close.

notice they are one second apart kinda impossible 

could be an issue with strokes for a 723 pump I guess - I was testing with a very small bolus as you can see

{ programmed: 0.025,
    _type: 'Bolus',
    _description: 'Bolus 2016-04-11T11:18:52 head[8], body[0] op[0x01]',
    timestamp: '2016-04-11T11:18:52',
    _body: '',
    _head: '0100010001000100',
    amount: 0.025,
    unabsorbed: 0.025,
    duration: 0,
    type: 'normal',
    _date: '74122b4b10' },
  { programmed: 0.025,
    _type: 'Bolus',
    _description: 'Bolus 2016-04-11T11:18:51 head[8], body[0] op[0x01]',
    timestamp: '2016-04-11T11:18:51',
    _body: '',
    _head: '0100010001000000',
    amount: 0.025,
    unabsorbed: 0,
    duration: 0,
    type: 'normal',
    _date: '73122b4b10' },

bewest, 2016-04-12:
Is this using mmeowlink+TI stick or regular, plain old carelink?

kenstack, 2016-04-12:
carelink