bewest/decoding-carelink #183: no cgm data II

efbest, 2016-04-24:
Frank_openaps $ mm-send-comm.py tweak ReadCurGlucosePageNumber
## do stuff with an insulin pump over RF
using ` Namespace(autoinit=False, begin=None, bytesPerRecord=None, command='tweak', descr=None, dryrun=False, effectTime=None, end=None, init=False, maxRecords=None, name=None, no_postlude=False, no_prelude=False, no_rf_prelude=False, other='ReadCurGlucosePageNumber', page=None, params=None, port='', postfix=None, prefix=None, prefix_path='', save=False, saveall=False, serial='', session_life=10, verbose=None) `
```
```
```
CRITICAL:decocare.stick:FAILED TO DOWNLOAD ANYTHING, after 0  expected:64
ERROR:decocare.stick:ACK is zero bytes!
Traceback (most recent call last):
  File "/usr/local/bin/mm-send-comm.py", line 4, in <module>
    __import__('pkg_resources').run_script('decocare==0.0.21', 'mm-send-comm.py')
  File "/usr/local/lib/python2.7/dist-packages/setuptools-20.3.1-py2.7.egg/pkg_resources/__init__.py", line 726, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/local/lib/python2.7/dist-packages/setuptools-20.3.1-py2.7.egg/pkg_resources/__init__.py", line 1484, in run_script
    exec(code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/decocare-0.0.21-py2.7.egg/EGG-INFO/scripts/mm-send-comm.py", line 8, in <module>
    app.run(None)
  File "/usr/local/lib/python2.7/dist-packages/decocare-0.0.21-py2.7.egg/decocare/helpers/cli.py", line 113, in run
    self.prelude(args)
  File "/usr/local/lib/python2.7/dist-packages/decocare-0.0.21-py2.7.egg/decocare/helpers/cli.py", line 130, in prelude
    uart.open( )
  File "/usr/local/lib/python2.7/dist-packages/decocare-0.0.21-py2.7.egg/decocare/stick.py", line 869, in open
    log.info('%s' % self.product_info( ))
  File "/usr/local/lib/python2.7/dist-packages/decocare-0.0.21-py2.7.egg/decocare/stick.py", line 510, in product_info
    return self.query(ProductInfo)
  File "/usr/local/lib/python2.7/dist-packages/decocare-0.0.21-py2.7.egg/decocare/stick.py", line 504, in query
    return self.process( )
  File "/usr/local/lib/python2.7/dist-packages/decocare-0.0.21-py2.7.egg/decocare/stick.py", line 491, in process
    ack, response = self.command.respond(raw)
  File "/usr/local/lib/python2.7/dist-packages/decocare-0.0.21-py2.7.egg/decocare/stick.py", line 66, in respond
    raise AckError("ACK is 0 bytes:\n%s" % lib.hexdump(raw))
decocare.errors.AckError: ACK is 0 bytes:

pi@raspberrypi ~/Frank_openaps $

bewest, 2016-04-25:
Linking these together: https://github.com/bewest/decoding-carelink/issues/182

bewest, 2016-04-25:
Apologize that the `mm-send-comm.py` interface is not as nice as openaps: in this case you need to add `--serial 123456` with your serial number.  You might also need to add `--init`; openaps itself tracks a lot of this info for you.

If this proves difficult, please file a feature request on openaps called "enable capturing raw data".