bewest/decoding-carelink #196: decoding page should detect crc

bewest, 2016-06-22:
Sometimes the decoder tries to eat the crc at the end as a record.

bewest, 2016-06-22:
```
pump://JSON/iter_pump_hours/settings/pumphistory-24h.json
Traceback (most recent call last):
  File "/usr/local/bin/openaps-report", line 82, in <module>
    app( )
  File "/usr/local/lib/python2.7/dist-packages/openaps/cli/__init__.py", line 51, in __call__
    self.run(self.args)
  File "/usr/local/bin/openaps-report", line 75, in run
    output = app(args, self)
  File "/usr/local/lib/python2.7/dist-packages/openaps/cli/subcommand.py", line 52, in __call__
    return self.method.main(args, app)
  File "/usr/local/lib/python2.7/dist-packages/openaps/reports/invoke.py", line 40, in main
    output = task.method(args, app)
  File "/usr/local/lib/python2.7/dist-packages/openaps/uses/use.py", line 45, in __call__
    output = self.main(args, app)
  File "/usr/local/lib/python2.7/dist-packages/openaps/vendors/medtronic.py", line 639, in main
    for rec in self.range( ):
  File "/usr/local/lib/python2.7/dist-packages/decocare/models/__init__.py", line 67, in __call__
    for record in page:
  File "/usr/local/lib/python2.7/dist-packages/decocare/models/__init__.py", line 46, in download_page
    for record in self.find_records(page):
  File "/usr/local/lib/python2.7/dist-packages/decocare/models/__init__.py", line 157, in find_records
    decoder = history.HistoryPage(response.data, self.inst)
  File "/usr/local/lib/python2.7/dist-packages/decocare/history.py", line 777, in __init__
    if lib.BangInt(crc) != computed:
  File "/usr/local/lib/python2.7/dist-packages/decocare/lib.py", line 235, in BangInt
    ( x, y ) = ints
ValueError: need more than 0 values to unpack

```

bewest, 2016-06-22:
Handled in PR: https://github.com/openaps/decocare/pull/1

loudnate, 2016-06-22:
> Handled in PR: openaps/decocare#1

Oh, you moved it?

bewest, 2016-06-22:
I did move it over to a place where it doesn't depend on my name, in particular.  I'd like to be able to consider roles in the diabetes industry, and in general this felt like a way to exercise a bit more discipline/rigor as a shared resource, or at least, that's the hope.